ArjanteMarvelde/uSDR-pico #2: Problem with waveshare LCD1602 RGB Module

gdelpech, 2022-01-04:
Hi Arjan

I have some problems with this display. I changed lcd.c and lcd.h and I rewrote these files with the instructions given on the Waveshare site and there it works except the special characters. I continue to seek to reinstate your two LCD files.
  I have started welding the plates but I am waiting for components.
However my LCD seems to be the same as yours.
[image]
[image]

Happy new year Arjan
73 QRO
Gilles F1BFU

ArjanteMarvelde, 2022-01-05:
Hi Gilles,

Happy new year to you as well! I hope we quickly forget 2021...

From the special character it seems you have an older lcd file. I think I solved some issues in the last version.
I your display a Seeed type?

. . . Arjan

ArjanteMarvelde, 2022-01-05:
Looking at your LCD it seems to be a different one than I use, probably a version with PCF8574 as I2C expander as there are many others like that. Maybe you can insert your (working) code bits into the latest v2.06 lcd.c of the repo?

gdelpech, 2022-01-05:
Hi Arjan

I tried with the latest version.
it is now just a backlight problem. The data is displayed correctly but very pale.

I will eventually find

73 Gilles F1BFU

ArjanteMarvelde, 2022-01-05:
Some displays reserve a bit for backlight on/off, but the exact bit may vary between types. 
I'm not sure how to handle all these variations...

closed 2022-02-25 by ArjanteMarvelde