ArjanteMarvelde/uSDR-pico #3: Using the LSD indicator with the PCF 8574 expander

ra9cua, 2022-02-06:
Hello.
My name is Sergey,I am in Russia.
I did not find such an indicator as yours and changed the code in the lcd.c  file a little.
I used a standard indicator with an i2s expander on PCF8574T
I don't know how to upload files here, if you are interested I can email them to you.

uint32_t hmi_step[6] = {10000000, 1000000, 100000, 10000, 1000, 100};	// Frequency digit increments 
                                             ?
hmi_option = (hmi_option<6)?hmi_option+1:6;					// Digit to the right

It was out of range.
I just added an even smaller rebuild step.
uint32_t hmi_step[7] = {10000000, 1000000, 100000, 10000, 1000, 100,50};	// Frequency digit increments 

I completely assembled the layout and noticed a noise at the output of the low frequency amplifier.
There are clicks in the loudspeaker.
 Sorry for my English, I use a computer translator. My email, if you are interested, I can send codes, oscillograms and my circuits.
My email is  ra9cua@mail.ru

73 SK RA9CUA  Sergey.

ArjanteMarvelde, 2022-04-25:
Thanks for the catch, it's been a while since I looked, but I will correct it in the code.

ArjanteMarvelde, 2022-08-12:
Closed, we're several version further now.

closed 2022-08-12 by ArjanteMarvelde

ra9cua, 2022-10-11:
Hello ArjanteMarvelde.
 
Thank you so much for not forgetting about me.I'm continuing your project.
I tried versions with b1 and b2 and I didn 't really notice the difference .
You did not plan to use the CW mode, in the first version there was a stub instead of the code.
This mode can be used to set up an automatic antenna tuner.He needs a continuous signal.

> 
> 
>Thanks for the catch, it's been a while since I looked, but I will correct it in the code.