ArjanteMarvelde/uSDR-pico #5: Still fighting the I2C LCD display

w9ran, 2022-02-23:
I made some changes and wanted to go back to the official branch where lcd.c 2.06 now includes the ability to use several display types.  I set the address #define to 0x27 (confirmed by running I2C scanner on an Arduino) and selected LCD_8574 GEN  as I'm using the cheap generic type of I2C LCD display.   The code is running as I can run commands via the serial interface, but the display remains blank.

I didn't see any other changes in the code that were required but maybe I'm missing something.  I also tried the LCD_8574_ADA but it doesn't work either.

I've spent more time by far fighting to get this silly display working than on the balance of the project ;-)   If anyone has a version of lcd.c that works with the generic display that uses the 8574 I2C daughterboard, I would very much appreciate receiving a copy so I can get on with more interesting things.   I did have it working using some code from another experimenter but would like to use the main branch if it can be made to work.   

Thanks and 73,
Bob W9RAN

ArjanteMarvelde, 2022-02-25:
Hi Bob,

Do you have a schematic of your display/I2C bridge combo? The crux is ultimately in how the 8574 pins are connected to the LCD driver. 
It is really sad that these modules are so badly standardized...

Arjan

w9ran, 2022-02-25:
Hi Arjan, and thanks for the response - I hate to keep beating this same dead horse ;-)
I traced out the 16 LCD pins on the units I have and then went online and compared with those I found.   Here is the schematic:
*
[image]
**
So this is clearly using a 4-bit mode, which I assume means I need to comment out all the other #defines in that group, so I tried that but still no-go.    I've compared with the LiquidCrystalI2C.cpp library for Arduino but about all I determined is that the display wakes up in 8 bit mode and needs to be put into 4 bit mode for this 8574 board to work.

I'm guessing that more parameters need to be configured and was hoping someone might have already figured out which ones.   I did have it working using a modified lcd.c provided by someone and if I could get that back I'd just call it done.

Thanks and 73,
Bob W9RAN

ArjanteMarvelde, 2022-02-25:
Bob, did you install jp4 (enable backlight) and do you have a contrast
cntrol pot?

Other than that it should work as a GEN variant.
I'll check the pinning and required bitmapping to be sure.

w9ran, 2022-02-25:
Yes, and the issue is partially resolved, and while I feel like an idiot at least I'm not the idiot who designed the protoboard I'm using.  I thought it looked nice, with a row of solder pads for each pin on the Pico to make it easy to solder up prototypes that are more reliable than solderless breadboards, but using the same basic design.  Except the guy who laid out this board forgot to put traces between the Pico and the rows of pads!   And I learned long ago that logic signals will not travel across a 5mm air-gap!
So now, after soldering on a bunch of jumper wires at least the display is doing something - flickering and showing a blinking cursor at top row position 3.    The backlight appears to be turning on/off many times per second, creating the flicker and no characters are visible at any setting of the contrast pot.   So something is not right but at least now I can see the effect of changes.
Running "lt" from the monitor now causes the display to blank then display a few random characters.   At this time the only changes in my lcd.c file are to enable the #define LCD_8574 and to change the I2C address to match my display.  If you have other suggestions I'd appreciate them and apologize for the noise caused by stupid protoboard assumptions.
73, Bob W9RAN

w9ran, 2022-02-25:
Additionally: 
I tried the LCD_8574_ADA hardware type even though I don't have the Adafruit display and with it I could see some bits of the display like the letters "USB" near the middle of the top line, although it was still flickering and continuously being overwritten or something,  basically unreadable.   Setting the display type to LDC_8574_GEN doesn't do anything but flicker, no characters are visible.    I should add that I've decreased the I2C data rate to 100kHz for the display.

w9ran, 2022-02-26:
Working now, I had a typo in one of the #defines.

closed 2022-02-26 by w9ran