ArjanteMarvelde/uSDR-pico #6: 20x4 lcd adress problem.

kefeur, 2022-04-22:
Welcome.
The problem is that I can't translate the code you wrote very well. I would ask you to send me a py file with the address 20x4 i2c lcd 0x27.
Thanks.

ArjanteMarvelde, 2022-04-25:
It is not a python implementation. 
It is also not designed for a 20x4 LCD. 
This SW is for experimenting, so if you cannot compile it, maybe you'd better first find out how that works? The howto is on the readme page.

kefeur, 2022-04-27:
Thanks for the reply, but we don't understand something.
what i found in the build folder i uploaded to my rpi pico card and the sound processing works.
Everything works on it except lcd handling.
I saw 2 * 16 on the ldc i2c port.
"Grove 16x2 LCD HD44780, with integrated JHD1804 I2C bridge (@ 0x3E)" I only have a problem with the title here!
All I asked was to change the address to the standard type 0x27.
If you have it please translate it and upload it to the build folder or send it just to me.
I would love to get it because I could give it a try.
What is described in the readme does not install on my computer at all.
Sorry
Please help me to try the result of your great work.
Thanks
Ferenc

ArjanteMarvelde, 2022-04-29:
And then I compile for the Grove display at 0x27, which is probably not going to work either. Before you know I am debugging your implementation and I don't have time for that.
Please try to install the SDK and development tools. Others have succeeded too...

kefeur, 2022-04-29:
Thank's

closed 2022-04-29 by kefeur