ArjanteMarvelde/uSDR-pico #7: core 0 stop running while receiving strong signal

agushart, 2022-05-08:
Thanks for your wonderful design work on Raspberry pi pico for sdr. Since Pico use 12 bit ADC, the sound is much better than the 10 bit ADC; and the RF noise is also low as your design use full balance mixer. However, I am still struggling with core 0 cut off when core 1 receiving strong radio signal. Core 1 still running even though core 0 stop working. I ‘ve checked the I - Q signal before ADC showing the signal voltage is still under 2 vpp which is less than 3.3 v (io vdd). Any clue to knowing the cause of the problem ?

ArjanteMarvelde, 2022-05-08:
I'm not sure what you mean with "core0 stop working". Did you check with the CLI on the serial port? Could it be the I2C interface?
Core1 processing is very loosely coupled with core0, in fact there is only some control signals between both. Also, a strong signal level could cause clipping of the signal, but I can't imagine how it would stop the processor.

Anyway, I have made some changes to the code that I will upload one of these days. Nothing functional, just some smoothing.

Apart from that I'm working on an FFT based signal processing engine, but this doesn't work as good as the current time-domain implementation.

agushart, 2022-05-09:
Thanks for your prompt response. Not only i2c stopped, the HMI also stopped as if the core 0 hanging. Anyway, I am very interested to check it thru the serial port stdio command shell.

ArjanteMarvelde, 2022-05-09:
Okay, well there could of course be a bug in the SW. If you find a deterministic way to reproduce it, please let me know.

GerritRCE, 2022-05-10:
Hi Arjan,
just made a copy of your usdx rp2040 repository.

Small note: 
If you set the PICO_SDK_PATH in your shell, and remove the "set PICO_SDK_PATH " from the CMAkelist.txt all persional settings are from the project.
This way it builds from git clone without any modifications.

Are you intending to maintain this project  

Nice project, just started with rp2040 and rf and software dev.   

ps, did you have anything to do with ERMES (or paging) projects in periode  ~1995 (ericsson)

Regards, 
Gerrit

agushart, 2022-05-11:
Sure sir

agushart, 2022-05-25:
The problem solved by changing DSP_US vector in line add_repeating_timer_us(-DSP_US, dsp_callback, NULL, &dsp_timer) of dsp.c subprogram.

closed 2022-05-25 by agushart