ArjanteMarvelde/uSDR-pico #10: About the 4.6 Audio module

fupingyi, 2022-08-15:
Hi, Arjan, 
Can you tell me the 4.6 Audio module in the schematic, where should the JP4 in it be connected? What do Line in and Line out in JP2 refer to?
Thanks!
73
Radio man

ArjanteMarvelde, 2022-08-15:
This is not well described, sorry. It just represents what I quickly whipped up on a vero-board, based on components I had on the shelf. For example, you wouldn't really use a 324 OpAmp for good audio buffering, better PSU decoupling, etc

Anyway: JP4 splits the 12V supply into 2x 6V. Make sure to NOT connect AGND and the general GND signals.
Line in and Line out on JP2 are exactly what it says it is. In fact it connects to the other audio module, the separation transformers that are implemented in an in-cable module. You need those to connect to the audio interface of a PC which runs the FT8 modem (or anything else).

The microphone input has a bias voltage to power the microphone. It has been made to comply with an ICOM mike, not sure about other.

fupingyi, 2022-08-16:
Sorry, I think maybe because I'm new to ham radio, I didn't get what you mean.
I guess I'll just wait for you to complete the circuit, provide a more complete schematic, and I'll do my research, the current schematic is a bit laborious for me.
In any case, thank you for your hard work and selfless work, and salute you.
73!

closed 2023-01-28 by ArjanteMarvelde