ArjanteMarvelde/uSDR-pico #12: Uninitialized variables in dsp_fft

howard0su, 2022-12-12:
I added more warnings to the compiler option, it catches the following ones. I believe there is code bug in the warnings:
/workspaces/uSDR-pico/dsp_fft.c: In function 'dsp_bandpass':
/workspaces/uSDR-pico/dsp_fft.c:138:45: warning: 'hi2' may be used uninitialized in this function [-Wmaybe-uninitialized]
  138 |  for (i=hi2+1; i<FFT_SIZE; i++) { XI_buf[i] = 0; XQ_buf[i] = 0; }
      |                                   ~~~~~~~~~~^~~
/workspaces/uSDR-pico/dsp_fft.c:137:45: warning: 'hi1' may be used uninitialized in this function [-Wmaybe-uninitialized]
  137 |  for (i=lo2+1; i<hi1; i++)      { XI_buf[i] = 0; XQ_buf[i] = 0; }
      |                                   ~~~~~~~~~~^~~
/workspaces/uSDR-pico/dsp_fft.c:125:14: warning: 'lo2' may be used uninitialized in this function [-Wmaybe-uninitialized]
  125 |  int i, lo1, lo2, hi1, hi2;
      |              ^~~
/workspaces/uSDR-pico/dsp_fft.c:125:9: warning: 'lo1' may be used uninitialized in this function [-Wmaybe-uninitialized]
  125 |  int i, lo1, lo2, hi1, hi2;
      |         ^~~
/workspaces/uSDR-pico/dsp_fft.c: In function 'tx':
/workspaces/uSDR-pico/dsp_fft.c:308:11: warning: 'ip' is used uninitialized in this function [-Wuninitialized]
  308 |  int16_t *ip, *qp, *ap, *xip, *xqp;

/workspaces/uSDR-pico/dsp_fft.c: In function 'tx':
/workspaces/uSDR-pico/dsp_fft.c:308:21: warning: variable 'ap' set but not used [-Wunused-but-set-variable]
  308 |  int16_t *ip, *qp, *ap, *xip, *xqp;

ArjanteMarvelde, 2022-12-14:
Thanks Howard, I will look into this when xmas holiday kicks in. One more day at work...

ArjanteMarvelde, 2022-12-14:
Meanwhile I checked and pulled in your changes, thanks.
Except the one about the single array element. I think it only saves a byte of storage, but it uses a different pattern than the other I2C cases...

howard0su, 2022-12-15:
single array element is just a code cleanup. I don't believe it brings any real gain. The syntax is a C trick that you can treat one variable as single element array when you get the address.

for the issues I mentioned in above warnings, it was addressed by #15 and the pending PR #19

closed 2024-07-06 by ArjanteMarvelde