ArjanteMarvelde/uSDR-pico #21: PCB Manufacture

n0th1n6, 2022-12-15:
Since I cannot find a way to ask questions about the PCB files, Ill just post it here even though it is not an issue. Is it worth to have the PCB professionally manufactured at this point? I don't have much luck working on perf board so I prefer a good quality PCB.

ArjanteMarvelde, 2022-12-18:
Well, for me these were protoboards manufactured by a prototyper (ant find the name, perhaps easyPCB or so?) for about $1 a piece.
I made the schematics with Eagle, but I think you can no longer download a free version. Of course there is always KiCAD if you want to make changes. Perfboard might indeed have issues with HF.

A package is downloadble from this git. If you need more, please let me know.

partal, 2023-01-27:
Hello.
I ordered the PCBs and completed the processor board and mixer board. The receiver is working fine.
I tried the modulation from the mixer output without microphone amplifier and power amplifier.
According to the diagram, the relays should be hfd3. But when I look at the hfd3 datasheet I think they will not fit the PCB
Do the relays have a different name? Or can you share a link where I can order?
I hope you can spend more time on the project.
Thanks
TA7MVL 73

ArjanteMarvelde, 2023-01-28:
Yes, the pcb layout is a close fit, but the relays I used are okay pinning-wise. 
I cannot get hold of my components or uSDRpico, since they're all in storage. I ordered from Reichelt, where from their page I think these are the ones I used.

TA1DB, 2023-02-01:
> Of course there is always KiCAD if you want to make changes

Hi @ArjanteMarvelde I could not import schematics neighter with KiCad version 5 nor Version 6. Could you please convert the project to KiCad if possible ?

ArjanteMarvelde, 2023-02-01:
Well, if you cannot import the files I cannot import them either. I don't know if import filters exist for this.

ArjanteMarvelde, 2024-07-06:
I just found this: https://community.element14.com/technologies/open-source-hardware/b/blog/posts/working-with-kicad-7-importing-eagle-files
Have not tried myself

TA1DB, 2024-07-06:
Thanks a lot, i'll check

ArjanteMarvelde, 2024-08-23:
Just a note, the schematics have been improved and ported to KiCAD (7) now.