ArjanteMarvelde/uSDR-pico #26: Restart Project

Trendyfu, 2024-07-06:
Hello, Arjan. Are you planning to restart the project and add waterfall to it?

ArjanteMarvelde, 2024-07-06:
Yes, I'm busy with a next version. The PCBs are about finished, save for some sanity checks and maybe some last minute changes.
Waterfall might be nice, but not a goal in itself. The uSDR-Pico design really is narrow-banded, so it wouldn't add a lot as such. Otherwise I would need to make the filtering more wideband, change the mixer and IF circuits, and have a whole new signal processing philosophy.
But you can take a look at Klaus Fensterseifer adaptations: https://github.com/kaefe64/Arduino_uSDX_Pico_FFT_Proj

Trendyfu, 2024-07-06:
OK, I understand. Good luck with your work, 73.

closed 2024-07-06 by Trendyfu