ArjanteMarvelde/uSDR-pico #28: build according pcb v4, but no sound.

firebat, 2025-01-12:
Hi Arjan, I build up a prototype according the v4 PCB schema.
[image]

 As logic analyzer shows, CPU-IO, RX and BPF works great. si5351 also seems works, but no signal was received. only some noise was heard. so what could be the problem?
[image]

For the TX board, when i plugin it into the bus, the power will be off.

ArjanteMarvelde, 2025-01-12:
Not sure how I can help? In fact I am also still testing my setup. 

One thing I did not yet test is the TX board, but I did plug it in and it does not disable the power…

If you only get RX noise, I would start to check whether the antenna is ok. If so, take it step by step from there, measuring all the levels and signals with a good oscilloscope.

Note that I had to make some patches on my PCB to make it work. This impacts especially the generation of the I-Q clock from the Si5351, and might be why you don’t receive anything. Please check the latest documentation for this.

 

Cheers,

Arjan

 

usdr.png (view on web) <https://github.com/user-attachments/assets/9578da15-b04b-47b3-8aa0-896b45c29806> 

i2c.png (view on web) <https://github.com/user-attachments/assets/4002fb00-4b67-48c4-be36-7e08e382a394>

ArjanteMarvelde, 2025-01-12:
Also, I don't think you can drive a loudspeaker directly from the line output.

firebat, 2025-01-14:
I have tested my antenna, and it worked well.
https://youtube.com/shorts/cj3ZNhzM7K4?si=UnpnADDUdFUOdoIu

and also modified the mix board following v4.1 patchs. (by replacing the uln2803a with uln2803lv, the hfd3/12 worked)
[image]

but still there is no radio sound, only some noise.
https://youtube.com/shorts/5a9QduOSgTU?si=V8NEMgWEfPNmrFQP

ArjanteMarvelde, 2025-01-14:
> [Tracking image][https://github.com/notifications/beacon/AG6P3I6PYMBRBB3QEK7MWKT2KRPLPA5CNFSM6AAAAABVBDIRS6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU2JH622.gif]

Is U4 pin 9 actually cut loose?

Do you have a scope to measure  I and Q clocks?

firebat, 2025-01-14:
I cut off U4 pin 8,  connected U4 pin 9 to pad 8 directly, which was connect to pin 12/13, and I have tested it with multimeter.
[image]

I and Q clocks looks like this on 7330 kHz, I'm not sure whether it's correct or not
[image]

ArjanteMarvelde, 2025-01-14:
Seems like the sampling rate of your scope /multimeter is too low? The clocks should be 50% duty cycle and 90 deg phase difference. A real scope is better, I think you should be sampling at least 8x clockrate or so.

firebat, 2025-01-15:
I setup the radio frequency to 1200khz
[image]

and sampling rate at 24Mhz, then I got the following graph. clock 1 was almost 180 deg delay 
[image]

and there were some breaks every 10ms.
[image]

ArjanteMarvelde, 2025-01-15:
> I setup the radio frequency to 1200khz
> 12.png (view on web)[https://github.com/user-attachments/assets/4e0eb37f-b955-43e5-942a-875f2dcb07ac]
>
> and sampling rate at 24Mhz, then I got the following graph. clock 1 was almost 180 deg delay
> iq.png (view on web)[https://github.com/user-attachments/assets/5399b65e-6923-444e-ad67-a734b185a145]
>
> and there were some breaks every 10ms.
> iq_break.png (view on web)[https://github.com/user-attachments/assets/e2f95eca-a47b-4c20-b12c-ddd696e132b9]

Don't go below 4MHz when testing, not sure what the lowest freq is. On the low side the phasing mechanism in the SI5351 stops working.

firebat, 2025-03-01:
Hi Arjan, I tested the clock with dso at 7050kHz, clock0 (yellow) is 90 degree delay than clock1(purple), is that OK?
[image]

ArjanteMarvelde, 2025-03-01:
Yes, the clk0 in SW is in fact a compound quadrature clock on the si5351 chip, using channels 0 and 1 in 90 degree shifted phases.
But you can change this as you want. Depends on your application.