ArjanteMarvelde/uSDR-pico #29: I didn't get tge

AntalVincz, 2025-02-01:
Hi 

I followed the build instructions but I did not get the uSDR.uf2 file in my build folder.

Here is the SDK build log

E:\uSDR-pico\build>cmake -G "Ninja" ..
PICO_SDK_PATH is C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/gcc-arm-none-eabi/bin/arm-none-eabi-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/gcc-arm-none-eabi/bin/arm-none-eabi-gcc.e
xe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/gcc-arm-none-eabi/bin/arm-none-eabi-g++
.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/python/python.exe (found version "3.9.13") found compone
nts: Interpreter
TinyUSB available at C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040;
 enabling build support for USB.
BTstack available at C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/btstack
cyw43-driver available at C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/lwip
mbedtls available at C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: E:/uSDR-pico/build

Thank you for your help

Regards

73' Tony
M0GLU

ArjanteMarvelde, 2025-02-03:
CMake just constructs the build environment for you. 
Then you still need to build the code itself by issuing "Ninja" in your build folder.

AntalVincz, 2025-02-03:
Can you please tell me how I can do it?
What is the command?

ArjanteMarvelde, 2025-02-03:
When you executed **_cmake -G "Ninja" .._** in the build folder, you instructed CMake to generate a build environment for the tool "Ninja".  
Hence this is the command you need to type in the CLI, just like the cmake, in the build folder.

Ninja is one of the alternatives for make.

romoloman, 2025-05-03:
There is a little mistake in hmi.cpp that prevent build from succeeding: at line 148 void hmi_nop(int){}; miss the variable name.

ArjanteMarvelde, 2025-05-04:
That's odd, I have compiled and tested the lot before uploading... 
Might be a different compiler setting?

Anyhow, I changed it. Thanks